madenvel/KalinkaPlayer #11: Improve the way volume correction is applied

madenvel, 2024-09-26:
Currently volume correction is applied to every track using the information provided by Qobuz. This works well for the mix of tracks from different albums but doesn't work if playing the same album with high dynamic range (i.g. Pink Floyd Dark Side of the Moon). This results in unwanted volume jumps between tracks which is not what was originally implied by the authors.

The feature is currently controlled via config file, so can't be easily changed (it also works for musiccast device, too).

A few suggestions here - whether make this feature easily configurable, or make it only switch if album changes.

alastair-dm, 2024-10-01:
The [ReplayGain spec](https://wiki.hydrogenaud.io/index.php?title=ReplayGain_2.0_specification) has tags for both track and album gain, as well as peak level for each. If the playlist has consecutive tracks from the same album then the album gain should probably be used instead of the track gain. LMS calls this SmartGain, but has a config option (per player) to use this, just track gain, just album gain or no replaygain. If you're not doing it already you should check the peak tag to check the gain doesn't push the peal over full scale.

madenvel, 2024-10-01:
Thanks @alastair-dm . Unfortunately, Qobuz doesn't seem to provide album gain information, only per track which is a challenge. Calculating it for the streamed tracks is a bit tricky.

The peak is not used as the way the gain is applied is only via volume control (currently on a musiccast device exclusively), so clipping is excluded.

The idea is to keep the data intact but operate with volume controls (alsa mixer in the future). This has both pros and cons ofc.

closed 2026-10-26 by madenvel